BingyaoHuang/single-shot-pro-cam-calib #2: 3d reconstruct

ll688ll, 2019-03-15:
Hi, thank you for sharing your code. I did calibrate the camera-projector system. Then could you please show how to use the calibration data to do the structure light reconstruct?

Thx

BingyaoHuang, 2019-03-15:
Reconstruction GUI will be released soon. For now, you can check [+Calibration/compareRealData.m](https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/b8045694511ba149e3c92edbb77567771f9edd31/%2BCalibration/compareRealData.m) for point cloud reconstruction from matched camera and projector point pairs. The structured light point pairs are stored in [nodesCell](https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/master/calibrateGUI.m#L101).

ll688ll, 2019-03-19:
> Reconstruction GUI will be released soon. For now, you can check [...]

Thank you Bingyao. The other issue I have is the camera. To get a better image quality, I have a Flir camera to set up the structured light system. I tried with your app. The camera can't be detected.  Is there any possibility to acquire images from Flir cameras?

Thank you.

BingyaoHuang, 2019-03-25:
FLIR cameras are not supported, because I use matlab's webcam interface, which only supports USB cameras. If your camera is point grey (acquired  by FLIR), a workaround is to use [MATLAB flycapture support](https://www.mathworks.com/hardware-support/point-grey-camera.html).

BingyaoHuang, 2019-04-03:
Reconstruction feature released. You may want to have a try.

closed 2020-04-01 by BingyaoHuang