BingyaoHuang/single-shot-pro-cam-calib #3: 无法打开calibAPP.mlapp文件

wenzhen-ye, 2019-03-22:
我下载压缩包后直接打开这个文件，却出现如下的错误，请问要怎么解决呢，是方法出现错误还是matlab版本的问题呢，我把错误贴出来，请您帮我解释一下，谢谢。
类 matlab.ui.container.Panel 不存在公共属性
Tooltip。
出错 calibApp/createComponents (line 893)
            app.CameraControlPanel.Tooltip =
            {'Control camera parameters. Tips:
            avoid camera overexposure by
            adjusting Exposure. Click Rest if
            you mess up with the parameters'};
出错 calibApp (line 1394)
            createComponents(app)

wenzhen-ye, 2019-03-22:
请问可以留下您的邮箱吗，我正在做跟结构光有关的三维重建，感觉您的项目对我很有帮助，想跟您请教一下问题可以吗。

wenzhen-ye, 2019-03-22:
请问mexopencv改如何使用呢，可以给详细一点的教程吗，谢谢啦。

BingyaoHuang, 2019-03-25:
mexopencv tutorial can be found at [mexopencv online document](http://kyamagu.github.io/mexopencv/) and [mexopencv github](https://github.com/kyamagu/mexopencv).

My email can be found in the paper.

ymk1933230135, 2020-03-25:
Have you solved this problem

ymk1933230135, 2020-04-01:
Hello, this problem has been solved, I use matlab2017 seems to not work, replaced by 2019.

closed 2020-04-01 by BingyaoHuang