BingyaoHuang/single-shot-pro-cam-calib #4: Error until calibration step

guangyuhe, 2019-03-24:
Hi, 

Before the step of calibration, every thing is fine. But when I click the calibration, the error appears:

Undefined variable "cv" or class "cv.FileStorage".

Error in Calibration.loadCalibInfo (line 29)
calibInfo = cv.FileStorage(fullfile(calibPath, 'calib-info.yml'));

Error in calibrateGUI (line 33)
calibInfo = Calibration.loadCalibInfo(fullfile(dataRoot, dataName));
Undefined variable "cv" or class "cv.FileStorage".

Error in Calibration.loadCalibInfo (line 29)
calibInfo = cv.FileStorage(fullfile(calibPath, 'calib-info.yml'));

Error in calibrateGUI (line 33)
calibInfo = Calibration.loadCalibInfo(fullfile(dataRoot, dataName));
Undefined variable "cv" or class "cv.FileStorage".

Error in Calibration.loadCalibInfo (line 29)
calibInfo = cv.FileStorage(fullfile(calibPath, 'calib-info.yml'));

Error in calibrateGUI (line 33)
calibInfo = Calibration.loadCalibInfo(fullfile(dataRoot, dataName));
Undefined variable "cv" or class "cv.FileStorage".

Error in Calibration.loadCalibInfo (line 29)
calibInfo = cv.FileStorage(fullfile(calibPath, 'calib-info.yml'));

Error in calibrateGUI (line 33)
calibInfo = Calibration.loadCalibInfo(fullfile(dataRoot, dataName));

Please advise. Thanks

guangyuhe, 2019-03-25:
The problem is fixed. Thanks anyway!!!

BingyaoHuang, 2019-03-25:
Good to hear that. Is this due to mexopencv installation?

guangyuhe, 2019-03-25:
Yes, it is. After the installation, I need to add the whole mexopencv folder to the path in Matlab.  Thanks

closed 2020-04-01 by BingyaoHuang

LuShixong, 2021-11-06:
so i must install the  mexopencv?

BingyaoHuang, 2021-11-06:
@LuShixong Yes.