BingyaoHuang/single-shot-pro-cam-calib #5: Error using snls (line 47) when running BA

ruoyuwangeel4930, 2019-06-30:
Performing camera-projector calibration using bundle adjustment (BA)...
**Error using snls (line 47)
Objective function is returning undefined values at initial point. lsqnonlin cannot continue.

Error in lsqncommon (line 164)
            snls(funfcn,xC,lb,ub,flags.verbosity,options,defaultopt,initVals.F,initVals.J,caller, ...

Error in lsqnonlin (line 252)
   lsqncommon(funfcn,xCurrent,lb,ub,options,defaultopt,allDefaultOpts,caller,...

Error in Calibration.stereoCalibrate (line 123)
param = lsqnonlin(objFun,param0,[],[],options);

Error in calibrateGUI (line 160)
stereoParams = Calibration.stereoCalibrate(Xm, Xc, Xp, calibInfo.camImgSize, calibInfo.prjImgSize,
'BA');

Error in calibApp/CalibrateButtonPushed (line 722)
                calibrateGUI(app); % call calibration function
 
Error using matlab.ui.control.internal.controller.ComponentController/executeUserCallback (line 335)
Error while evaluating Button PrivateButtonPushedFcn.**

ruoyuwangeel4930, 2019-06-30:
Is this related to the:
Warning: Color grid too dense, try moving the board away from the projector or moving the camera closer to the board 
Thanks for your reply!

BingyaoHuang, 2019-07-01:
I think it's related to the color grid density. Try moving the board away from the projector or moving the camera closer to the board.

closed 2020-04-01 by BingyaoHuang