BingyaoHuang/single-shot-pro-cam-calib #8: Warning: Image is too large to display on screen; will show up at 67%.The object is invalid or has been deleted.

xubin2015, 2019-09-10:
Thank you very much!
When using  your system, I met a problem: when I push the capture button,the screen is dark and promted the following problem:
Warning: Image is too large to display on screen; will show up at 67%.The object is invalid or has been deleted.
Can you give me a email? I share the video with you.

xubin2015, 2019-09-10:
I believe my pattern is right and I following your instruction in your video.
Best wishes to you.

BingyaoHuang, 2019-09-11:
Sure. bingyao dot huang AT temple dot edu.

xubin2015, 2019-09-11:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    
    
        Thank you very much.You can learn the problem I met by seeing the follwoing eight pictures. After I press the capture button,the screen are dark.Best wishes.
    
        
            
        
    
    
     
    
        
                    
            
                
                    
                            
                                
                            
                            
                                xu_bin
                            
                            
                                
                            
                    
                        
                            
                                    xu_bin@tju.edu.cn
                            
                        
                
            
        
        
    
    
        签名由
        网易邮箱大师
        定制

closed 2020-04-01 by BingyaoHuang