BingyaoHuang/single-shot-pro-cam-calib #10: Regarding the camera resolution

guangyuhe, 2020-07-19:
Hi,

The code runs very well when the camera's resolution is '640x480'. The success rate is almost 100%.  However, when changing to a higher camera's resolution, like '1600x1200', the success rate drops a lot.
 I do see your notes that it has a better performance when using a lower resolution.  But I still would like to ask if there is anything I can do to improve the behavior when using a higher camera resolution. And what could be the possible bottleneck for this problem? 

Would be appreciated for any suggestions!

BingyaoHuang, 2020-07-19:
@guangyuhe The reason is that the camera-captured color grid line thickness (in pixel) is changed by the camera resolution (1600x1200), thus the default color grid segmentation parameters may not work well.
 
For other resolutions, e.g., 1600x1200, you can modify the color grid segmentation parameters such as [adaptiveThresh](https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/master/%2BImgProc/adaptiveThresh.m) sigma, morphological operation's kernel sizes in [segColorGrid](https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/master/%2BImgProc/segColorGrid.m#L176) and [bw2skele](https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/master/%2BImgProc/bw2skele.m#L52) accordingly. To find the best parameters, you can turn on the `verbose` flag for more debug info and see if the color grid is correctly segmented.

guangyuhe, 2020-07-19:
Thank you. It is a very good suggestion. 

In segColorGrid[https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/master/%2BImgProc/segColorGrid.m#L176](url), I make the following change from line 175: 
```
% close and open
se1 = strel('line',10,0);%parameter for '1600x1200'
se2 = strel('line',10,90);
% se1 = strel('line',2,0);%original
% se2 = strel('line',2,90);%original
```

Now, the behavior is very good.

BingyaoHuang, 2020-07-20:
Great!

closed 2020-07-31 by BingyaoHuang