BingyaoHuang/single-shot-pro-cam-calib #13: 报错:未定义"cv"或“cv.filestorage”

LuShixong, 2021-11-07:
未定义变量"cv" 或类"cv.FileStorage" 。
出错Calibration. loadCalibInfo (line 29)
calibInfo = cv.FileStorage( fullfile(calibPath，'calib-info.yn .
出错calibrateGUI (line 31)
calibInfo = Calibration. loadCalibInfo(fu11file(dataRoot，data )
我们安装了mexopencv，但是仍然解决不了这个问题
可以回复一下吗？

BingyaoHuang, 2021-11-07:
Seems mexopencv is not added to MATLAB path. 

Execute this in MATLAB command line:
``addpath('C:\path\to\mexopencv')``

Then execute this in MATLAB command line:
``help cv``

LuShixong, 2021-11-08:
after we failed,we choose the .EXE file ,we setup the MATLAB runtime 2021a in the path "matlab/toolbox\compiler\deploy\win64",we try again,but we still failed, so what's wrong?
P.S.: we ues the photo in your file that i downloaded,but we cannot figure it out

BingyaoHuang, 2021-11-08:
Can you post the error message?

LuShixong, 2021-11-09:
Of course,
we set up the opencv3.4.1 and mexopencv in the MATLAB file and 
addpath('G:to\mexopencv\mexopencv-3.4.1\mexopencv-3.4.1')
and addpath('G:to\opencv3.4.1\opencv')
and  addpath('G:to\single-shot-pro-cam-calib-0.1')
but that's the result:
[image]
i don't know what's wrong with it. i tried 2019a and 2021a and .EXE ,but they all failed. 
when i clicked the calibrate button in the .EXE ,nothing happened.
i really want to learn more about

BingyaoHuang, 2021-11-09:
Maybe it's due to mexopencv version (you installed 3.4.1). However, in README usage section, I mentioned using mexOpenCV v3.3.0.

Try mexOpenCV v3.3.0 and let me know if this error pops up again.

closed 2023-07-27 by BingyaoHuang