BingyaoHuang/single-shot-pro-cam-calib #14: Windows Executable - Stuck at "Extracting checkerboard corners from camera image..."

eldrickm, 2022-03-07:
Hi Bingyao,

I'm currently trying to use the windows executable and have captured a series of poses. Upon calibrating, I seem to be stuck at the above message with the progress bar not moving. I have tried manual segmentation masks and the captured images seem to look fine - both the plain and the projected grid images. I've attached a few below for verification.

What is the issue here?

I am on Windows 10 and am using the MATLAB R2021a runtime.

[image]
[image]

BingyaoHuang, 2022-03-08:
Can I see the message in `log.txt` under your installation directory?

eldrickm, 2022-03-08:
Hi Bingyao,

I solved this by using the exact number of checkerboard rows and columns as shown in your example images.

Thanks!

closed 2022-03-08 by eldrickm