BingyaoHuang/single-shot-pro-cam-calib #16: Windows Executable - Unable to calibrate

MathieuVDC1, 2022-03-10:
Hi, Everything works fine until I click "Calibrate", and then nothing happens.

Seems to be some file error when looking at the log file:

Error using FileStorage
Failed to open file

Error in Calibration.loadCalibInfo (line 29)

Error in calibrateGUI (line 31)

Error in calibApp/CalibrateButtonPushed (line 848)

Error in appdesigner.internal.service.AppManagementService/tryCallback (line 377)

Error in matlab.apps.AppBase>@(source,event)tryCallback(appdesigner.internal.service.AppManagementService.instance(),app,callback,requiresEventData,event) (line 37)

Error using matlab.ui.control.internal.controller.ComponentController/executeUserCallback (line 410)
Error while evaluating Button PrivateButtonPushedFcn.

Any idea how I could solve this? I'm using Windows 10 and 2021a Matlab runtime.

closed 2022-03-10 by MathieuVDC1

BingyaoHuang, 2022-03-11:
Great! Thank you for the feedback!

fare129, 2024-03-01:
Could u tell me what is the calib_info.yml? Why can't i find it, thanks