BingyaoHuang/single-shot-pro-cam-calib #17: Significant reprojection errors

517pacifikal, 2023-07-27:
Our calibration results suffer significant reprojection errors for projector and stereo ones when using this camera-projector calibration software. We have increased calibration samples to 15 to reduce the noise during capturing but still cannot solve the problem. After multiple attempts, we still cannot find a proper solution to calibrate our camera-projector system.

In our experiment setup, we use a 5x8 corner checkerboard for calibration and the length of each grid is 90mm.
The collected checkboard poses involve different orientations and distances.
We here post some sample images in our calibration set and hopefully get your valuable suggestions.
[image]
[image]
Besides, we list some minor questions below:
1. how to use a customized checkboard with different corner resolutions?
2. how to use pre-calibrated camera parameters to initialize the calibration process? Would it be possible to provide detailed instructions?
2. Is there any distance or posing rules that we need to follow during the image capture?
3. Our error reaches 100+ for projector and stereo calibration, but the image sample has a similar contrast and brightness as the example images in the repo. Could you please give us detailed suggestions on solving this issue?

Thank you very much.

BingyaoHuang, 2023-07-27:
1. We use MATLAB's [detectCheckerboardPoints](https://www.mathworks.com/help/vision/ref/detectcheckerboardpoints.html) to automatically detect the checkerboard, including its #corners.
2. Check this [line of code](https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/master/calibrateGUI.m#L85), and you may replace `camParams` with your pre-calibrated camera parameters.
3. The camera may be too far from the board, try moving the camera closer. 
4. For the data you emailed me, the board poses are similar, you can try different poses, e.g., 90 degrees rotation, larger translations, etc.

You can run the app with "Verbose (debug)" checked, and see whether the intermediate results look reasonable.

517pacifikal, 2023-08-08:
We've solved this issue by moving our camera closer to the board, now the reprojection error have reduced to an acceptable level.
Thank you for your help!

closed 2023-08-08 by 517pacifikal