BingyaoHuang/single-shot-pro-cam-calib #18: Obtain homography between camera and projector from calibration

RDKKKKKK, 2023-08-16:
Hi, thanks for the amazing software, the error is quite acceptable, really solved my trouble. But when I use the calibration result to calculate the homography between camera plane and projector plane, the outcome is not desired. I wonder is there an accurate way to obtain the homography from the calibration? 
Also I notice the nodePairs.yml in the output, I'm a little confused about the format in it.

BingyaoHuang, 2023-08-19:
The homography may be inaccurate if your projector or camera has significant lens distortions.
`nodePairs.yml` stores the color grid nodes' coordinates in projector and camera image spaces.

RDKKKKKK, 2023-08-21:
Thanks, i check the distortion but find it not significant. Maybe my method is not correct, currently i simply project some random 3D points onto camera and projector planes respectively, and use `cv2.findHomography` and `cv2.warpPerspective` to warp my camera image to projector view, may i ask is there a better way to do this?

BingyaoHuang, 2023-08-22:
It may work if both images/2D points are undistorted, and the random points are correctly matched.

closed 2023-08-25 by RDKKKKKK