BingyaoHuang/single-shot-pro-cam-calib #19: error use  cv.findHomography

fare129, 2024-03-02:
Hi Huang, now that I've run into this problem, can you please give some advice? I am using matlab2021a+mexopencv-3.4.1.
Detail：
错误使用 cv.findHomography
MEX �ļ���������ı�׼�쳣��
What() ��: OpenCV(3.4.1) D:\dev\opencv\modules\calib3d\src\fundam.cpp:372: error: (-5) The input arrays should be 2D or 3D point sets in function
cv::findHomography

��.

出错 Calibration.getNodesAndPrjCorners>findMatchAndWarp (第 83 行)
[Hcp, inliers] = cv.findHomography(camNodesUndistort, prjNodes, ...

出错 Calibration.getNodesAndPrjCorners (第 47 行)
        [prjCorners(:,:,i), nodePairs{i}] = findMatchAndWarp(imLightNames{i}, ...

出错 calibrateGUI (第 93 行)
[nodesCell, ~] = Calibration.getNodesAndPrjCorners(calibInfo, camParams, camCorners, verbose, useParallel);

出错 calibApp/CalibrateButtonPushed (第 868 行)
                calibrateGUI(app); % call calibration function
 
错误使用 matlab.ui.control.internal.controller.ComponentController/executeUserCallback (第 410 行)
计算 Button PrivateButtonPushedFcn 时出错。

fare129, 2024-03-02:
These are the photos I took.
[image]
[image]
[image]
[image]
[image]
[image]

fare129, 2024-03-02:
These are the photos I took.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

BingyaoHuang, 2024-03-04:
Can you check whether any of input to `cv.findHomography` is empty?

fare129, 2024-03-04:
yes, it is empty, and when I use your data, the program works fine, but the data I collect myself reports an error. Can you tell me if it is because the background of the picture taken has an effect? A pure white background is needed is it?

BingyaoHuang, 2024-03-04:
Maybe the board segmentation did not work well on your setup. You can set break point and debug the code, see whether the intermediate results look reasonable. You can tick `Verbose` option in the software to enter debug mode.

fare129, 2024-03-04:
Yes, it will report an error on the segmentation part, does this look like it means that the background should be as whiteboard background as possible, as much as possible as in your case?
[image]

fare129, 2024-03-04:
Here's a photo captured from my second experiment.I don't think the interference should count for much anymore, but it's still a mistake in the segmentation.
[image]
[image]

BingyaoHuang, 2024-03-04:
> Yes, it will report an error on the segmentation part, does this look like it means that the background should be as whiteboard background as possible, as much as possible as in your case? ![1709543917768](https://private-user-images.githubusercontent.com/85798880/309698275-c17526ab-ef91-4720-bb05-8ea088bc6a92.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk1NDcwNzksIm5iZiI6MTcwOTU0Njc3OSwicGF0aCI6Ii84NTc5ODg4MC8zMDk2OTgyNzUtYzE3NTI2YWItZWY5MS00NzIwLWJiMDUtOGVhMDg4YmM2YTkyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzA0VDEwMDYxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg0NjQ1MTRhMmU2OWQ2NTA5YmFlODRiYzIwYmJhZWU4MTI0MjQyODFmZDg5NzY1NmZmMTA1Mzg1NDFlY2Y4NTEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0._8-HPMZmKhauSHNArIVw-A47fefLj7DfwbN8hRn7sUs)

The checkerboard region should be all black. You can check the segmentation results of my data, and compare it with yours.

closed 2024-03-31 by fare129