BingyaoHuang/single-shot-pro-cam-calib #20: The calibration program always stays on calibrating

yujiang0922, 2024-03-08:
when I tried with MATLAB Runtime way, the calibration program always stays on calibrating
[image]

is there anything wrong?

yujiang0922, 2024-03-08:
the error log:

Extracting checkerboard corners from camera image...
Calibrating camera using checkerboard corners...
Extracting and saving grid nodes and warped projector corners...
Starting parallel pool (parpool) using the 'local' profile ...
Connected to the parallel pool (number of workers: 8).
Extracting projector corners for image 1
Extracting projector corners for image 2
Extracting projector corners for image 3
错误使用 Calibration.getNodesAndPrjCorners>findMatchAndWarp (第 83 行)
MEX 文件 'C:\Users\jiang0922.yu\AppData\Local\Temp\jiang0922.yu\mcrCache9.10\ProCam0\Users\bingy\Documents\git\mexopencv\+cv\findHomography.mexw64' 无效: 找不到指定的模块。

出错 Calibration.getNodesAndPrjCorners (第 52 行)

出错 calibrateGUI (第 93 行)

出错 calibApp/CalibrateButtonPushed (第 848 行)

出错 appdesigner.internal.service.AppManagementService/tryCallback (第 377 行)

出错 matlab.apps.AppBase>@(source,event)tryCallback(appdesigner.internal.service.AppManagementService.instance(),app,callback,requiresEventData,event) (第 37 行)

错误使用 matlab.ui.control.internal.controller.ComponentController/executeUserCallback (第 410 行)
计算 Button PrivateButtonPushedFcn 时出错。

BingyaoHuang, 2024-03-10:
Check whether `findHomography.mexw64` is under that directory.

closed 2024-11-23 by BingyaoHuang

TIMESTICKING, 2025-01-14:
It's under that folder, then what to do. I'm using windows, and executing it from zip file

BingyaoHuang, 2025-01-15:
I need more info

TIMESTICKING, 2025-01-15:
It's the problem of "parallel", when I uncheck "parallel", the error gone

BingyaoHuang, 2025-01-15:
Great, thank you for sharing this.