BingyaoHuang/single-shot-pro-cam-calib #21: Here are some problem about "Objective function is returning undefined values at initial point. lsqnonlin cannot continue."

fare129, 2024-04-03:
Hello, thank you very much for the program, now I ran into a problem, is that I think I took the photo is very clear, why always encounter the error "Objective function is returning undefined values at initial point. lsqnonlin cannot continue“. And, according to the debug result, the error occurs during the process of fixing the point. How to solve this problem? Very much looking forward to your reply.
[image]
[image]
[image]

BingyaoHuang, 2024-04-03:
The color grid is not well decoded, the blue and red stripes are washed out. Try adjusting your projector and camera color settings.

fare129, 2024-04-03:
Thanks for your respond, and my projector and camera both have a resolution of 1280 X 720.. Could it be a resolution issue, please?

fare129, 2024-04-04:
The problem has been solved, firstly because the sports camera used, with wide angle shots, results in poor image quality. Secondly because the projector can't be too close to the calibration plate, otherwise the stripes are not clear from each other. After changing the camera and distancing the projector, the problem was solved. Thanks again.

closed 2024-04-04 by fare129