BingyaoHuang/single-shot-pro-cam-calib #22: Rotation matrix and translation matrix problem

fare129, 2024-04-27:
Hello, thank you again for your help. Now I find two thorny questions:
 1. After I calibrate according to the requirements, is the translation matrix given arranged according to [X, Y, Z]? This is obviously inconsistent with the position map obtained by the operation. From the given map, it can be clearly seen that if displayed according to [X, Y, Z], the translation matrix should be [416.707,     -129.316,      246.134], Instead of the [246.134,     -129.316,      416.707] you gave.
 2. If the translation matrix is not correct, is the rotation order of the rotation matrix also incorrect? I looked at the source code and saw the rotation sequence of XYZ. Is this the final rotation sequence? If not, so what? Please let me know.
Looking forward to your reply
[image]
[image]
[image]
[image]

BingyaoHuang, 2024-04-27:
Translation vector (T) and the location of the projector (prjOrg) are different things. https://github.com/BingyaoHuang/single-shot-pro-cam-calib/blob/cd7fda6b98d86175ccb4a5a0669998f311c55b00/%2BReconstruct/visualizePts3d.m#L30

fare129, 2024-04-27:
Okay, what about the rotation matrix? Is it rotated in the XYZ direction on the translation vector, or in the reprojected XYZ direction? Or some other rotation sequence.

BingyaoHuang, 2024-04-27:
It follows OpenCV coordinate system and aims to rotate a point from camera view to projector view space.

fare129, 2024-04-27:
What I want to say is that according to the results, the translation matrix is the camera translation amount obtained by taking the projector as the origin of the coordinates. So the rotation matrix should also be the rotation of the camera coordinate system relative to the projector coordinate system, right? If this is the case, I would like to ask, what is the rotation sequence? Because the rotation order is different, the restored coordinate system coordinates are different, which is very important to me. Could you please let me know?