BingyaoHuang/single-shot-pro-cam-calib #24: Camera Light Image and Camera Grid Image are always the same

ENKRAN, 2024-11-23:
Hello,

currently i'm trying to capture images for different sets but 99% of the time, the Camera Light Image and the Camera Grid Image are the same and the grid is visible in both images. I tried many different things like adjusting settings in the camera control, rearranging the projector and camera, changing resoulutions, try to copy almost everything from your tutorial video and description but it still doesn't work like it did in your video on youtube. On very rare occations the capturing works for one set even though i didn't change any settings or lighting in the environment at all so i really can't tell what's causing this problem. An example screenshot is
attached too: 
[image]

Help in some kind of way would be really appreciated. Thanks in advance.

BingyaoHuang, 2024-11-23:
It's caused by the camera delay, and different cameras may have different delay times. 
Try increasing Exposure Time (s) to more than 2 seconds in the MATLAB code and let me know.

ENKRAN, 2024-11-24:
Thanks for the fast reply! That worked, thank you.

closed 2024-11-24 by ENKRAN